bruceborrett/easy-enclosure #2: Transparent cover - which filament?

ser, 2023-09-08:
Hello,
I am sorry if I'm abusing issues but are you able to recommend from your experience a filament which would make a proper transparent or semi-transparent cover to see the content of the box?

Thanks!

bruceborrett, 2023-09-08:
Im afraid I do not have any experience with transparent filament, perhaps someone else will have some advice on this then I can add a note to the readme.

bruceborrett, 2023-09-12:
I found this great article which hopefully solves this question:

https://www.3dsourced.com/rigid-ink/clear-3d-printer-filament/

closed 2023-09-12 by bruceborrett

ser, 2023-09-12:
A really nice explanation!

"While it’s impossible to print 100% clear models, printing simple designs with thin walls will certainly help."

Are we able to select how thin is the cover in the app?

bruceborrett, 2023-09-14:
I have made some updates and it is now also possible to configure the thickness of the lid.