bruceborrett/easy-enclosure #4: feature: improve UI around multiple holes/pcb mounts

erikarenhill, 2023-09-14:
I'm suggesting to build the sidebar/UI around a collapsible accordion, where you modify the amount of holes/pcb within that accordion by putting an "add" and "remove" button for each hole. The purpose with accordion is that you can collapse the "holes" section while working on the PCB mounts.

The purpose with add/remove button is that it will enable the possibility to individually select which hole/mount to remove instead of just changing the quantity and it would also open up for avoid reseting the other. E.g. now when you change the amount the settings are reset and you have to re-do all configuration for it.

erikarenhill, 2023-09-14:
I'd love to take a stab at this and make a PR for it

erikarenhill, 2023-09-15:
> This is exactly what I had in mind and yes you may absolutely give it a go. I have just pushed a couple changes to the form so please sync your repo (if already forked) before starting.

I have started to do some work on this yesterday at https://github.com/bruceborrett/easy-enclosure/compare/main...erikarenhill:easy-enclosure:refactor/parameterform (work in progress).

I just want to discuss adding another dependency before proceeding

I'm thinking that introducing a form library, specifically React Hook Forms that would make a bit much easier,
it would allow to get rid of all the state params handling throughout the code and also give some features for free such as adding/removing arrays (useFieldArrays) (multiple holes/pcb) without having to manually manage these through states.

What's your opinion about that approach @bruceborrett ?

bruceborrett, 2023-09-15:
Yes a form library is probably the way to go, Im most familiar with Formik but open to whatever. 

I played around a bit yesterday with using onblur event rather then onchange along with a keydown handler to blur the input when pressing enter and it makes UX a million times better and solves the laggy constant re-rendering issues.

I will also be adding a loading indicator to show when the model is busy re-rendering (this will be separate from the form and so wont interfere with what you doing).

erikarenhill, 2023-09-15:
I have no experience in Formik, but I can do a proof of concept using React hook forms so we have something to decide upon, I don't want to introduce a dependency you're not comfortable with as you are the project owner, but I think RHF is really nice

erikarenhill, 2023-09-15:
Just had a quick look at Formik, and it looks very similiar to RHF in the approach, so I can try to do it with Formik instead, and get some experience with another form library on the way

erikarenhill, 2023-09-21:
I had a rough week in terms of lack of time, but just wanted to let you know I have not forgotten about this. :) If you want to progess without my contribution feel free to, otherwise I will do this when I find time, as soon as possible!

bruceborrett, 2023-10-06:
@erikarenhill just a heads up that I have been working on this and have gotten quite far, hoping to push in the next couple days.

erikarenhill, 2023-10-06:
> @erikarenhill just a heads up that I have been working on this and have gotten quite far, hoping to push in the next couple days.

Thanks for heads up! Ive been really busy lately so sorry for not pushing this further. Looking forward to your update and maybe I can contribute from there instead

bruceborrett, 2023-10-09:
@erikarenhill I have pushed my changes, please give it a test and let me know if you find any issues

erikarenhill, 2023-10-09:
> @erikarenhill I have pushed my changes, please give it a test and let me know if you find any issues

Thanks for the update, I found a couple of things when running the latest main-branch locally:

If you delete holes, eventually the app will crash with "Uncaught runtime errors: wrong number of arguments", just click delete on hole 1 until it crashes.

The labels and input doesn't seem to be correctly linked (try and click the "Waterproof" text next to the checkbox, and the checkbox should be toggled), the same applies to scews and mounts.

Personally I'd like to have the possibility to have multiple accordions open at the same time instead of having them autocollapse when clicking another one, so you could have the ones open that you need/would like to work with.

bruceborrett, 2023-10-13:
>If you delete holes, eventually the app will crash with "Uncaught runtime errors: wrong number of arguments", just click delete on hole 1 until it crashes.

This is now resolved by https://github.com/bruceborrett/easy-enclosure/commit/c5017ae5d9268a747309037c9581be29ccd12642

>The labels and input doesn't seem to be correctly linked (try and click the "Waterproof" text next to the checkbox, and the checkbox should be toggled), the same applies to scews and mounts.

This is working fine on my end, however there is quite a delay after clicking the checkbox. The delay is due to the fact that the re-rendering process is currently not very efficient, I will be trying to optimise this in future.

bruceborrett, 2023-11-02:
The delay has now been resolved by https://github.com/bruceborrett/easy-enclosure/commit/4168fbf9929dc6ff7c7e7c2a21332f0ac61475b3. Closing this issue.

closed 2023-11-02 by bruceborrett