bruceborrett/easy-enclosure #14: Loading from file doesn't change settings

contagon, 2023-10-08:
First off, thanks for such an awesome tool! I'm going to be using it a lot.

I having some issues saving and then loading from file. Saving works fine, and when I load from file the box looks correct, but the settings don't change from what they were previously. This means I can't load and tweak a previous design.

For example, my design has a width of 110, but when loaded only shows 100 in the settings, but the visualized box does look correct:
[image]

Let me know if there's anything I can do to help troubleshoot. It appears to happen to all the settings.

buccaneer-jak, 2023-10-31:
I see the same problem. Saving and then importing does not then show the correct dimensions. But overall awesome.

bruceborrett, 2023-11-02:
This is now fixed by https://github.com/bruceborrett/easy-enclosure/commit/64ff961cee2c7b0d3642e07d36102467b25388ff

closed 2023-11-02 by bruceborrett