bruceborrett/easy-enclosure #16: Saved File can't be opened

BigJoeFour, 2023-10-09:
Previous design that was saved can no longer be opened to make changes to. Attached file renamed from json to txt to allow uploading.
[enclosure-2023109165011.txt](https://github.com/bruceborrett/easy-enclosure/files/12847728/enclosure-2023109165011.txt)

contagon, 2023-10-09:
I had the same problem, and I think this is due to an update and how the json files are exported. Here's a current one that does import correctly (changed to .txt as well):
[enclosure-v3.txt](https://github.com/bruceborrett/easy-enclosure/files/12848848/enclosure-v3.txt)

bruceborrett, 2023-10-11:
Yes unfortunately since this software is still in early stages, the settings json structure will probably still change quite often and it will be difficult to keep backwards compatibility with settings that were exported earlier.

Going forward I will try  my best to minimise the impact of these changes and keep backwards compatibility where possible.

closed 2023-10-11 by bruceborrett