bruceborrett/easy-enclosure #17: great tool - issues and new suggestions

macnug, 2023-10-31:
Hi, this tool in awesome, nothing comparable online!

I tested the tool and tried to print some enclosures for a esp8266 + VL53l1x people counter project.
I figured out some issues:
- printing in petg i have tons of stringing and i didn't succed in having it fixed...
- while importing the stl in cura, i had always a message saying that the stl is incomplete, is something regarding the stl creation?
- the pcb mount if high are very fragile, would be nice to have some reinforcement on that.
- last, a mm grid on back of the model would be very useful in order to design in a better way the models.

Thank you!!
Marco

bruceborrett, 2023-11-01:
Thanks for the great suggestions and feedback Marco!

Please see my comments below:

> printing in petg i have tons of stringing and i didn't succed in having it fixed

So do you mean you do not experience any stringing with other models you have printed? I have found this to be a general issue with PETG and Ive had to spend a lot of time fine tuning my settings to minimise stringing, furthermore, Ive found that every brand of filament requires different settings

> while importing the stl in cura, i had always a message saying that the stl is incomplete

Ive not experienced this myself and I also use Cura, did you perhaps have holes that were overlapping with the waterproof seal or something to that effect?

> the pcb mount if high are very fragile, would be nice to have some reinforcement on that

Yes this needs to be addressed, I will create a separate issue for this

> a mm grid on back of the model would be very useful in order to design in a better way the models

Yes many people have raised this, I will also create an issue just for this

> would be great to have the possibility to have pcb mounts also on the lid

Yes why not, will create an issue for this as well

bruceborrett, 2023-11-17:
Closing this for now, feel free to re-open if you have any further feedback.

closed 2023-11-17 by bruceborrett