bruceborrett/easy-enclosure #22: Importing JSON does not update values on screen.

Jmoore4717, 2023-11-16:
When I import a model I have previously created with this the values do not update on the right side of the screen. So if I am trying to modify the enclosure i have to open the JSON in a text editor to manually copy all the values over. This makes making changes a bit more time consuming and tedious.

bruceborrett, 2023-11-17:
Ive just tested loading a JSON file and all values in the form do update correctly. Im probably just misunderstanding your description of the problem but I cant see why you are needing to edit the JSON file?

Jmoore4717, 2023-11-17:
Ill try to explain it better. So i created an enclosure and saved the json file. I printed it and realized i need the enclosure a little bit bigger. So i uploaded the JSON file to the easy enclosure page. It loads fine. But the values on the right do not update with the values that were uploaded. So if it try to make a change, the enclosure reverts back to the default parameters because that is what is in the values on the right. Making small changes to the enclosure is difficult because i have to open the JSON and copy the values to the right.

Btw this is a great tool and even as it is, it is very helpful.

Jmoore4717, 2023-11-17:
Hmmm actually i just tried this again and seems to be working. Maybe it was an issue with my browser but it was not updating the values before. The first time i tried this and tried to change a value, the enclosure reverted back to the default values and i had to start all over configuring the enclosure. 

Anyway thank you for looking at this, i appreciate your time.

bruceborrett, 2023-11-17:
@Jmoore4717 could you please upload the json file that you are trying to load?

Jmoore4717, 2023-11-17:
Here it is. It is working now though. Not sure why it didn't before.

[enclosure-20231021111311.json](https://github.com/bruceborrett/easy-enclosure/files/13396867/enclosure-20231021111311.json)

closed 2023-11-29 by bruceborrett