bruceborrett/easy-enclosure #28: Multiple designs where Lid does not fit

jasonspo, 2024-10-12:
I have had three designs where the lid would not fix the bottom due to the insert conflicting
 with the fill areas near the screw holes.   I've had to resort to remove the insert on the design for the lid to close.

attached is json where the problem is occurring.  
[enclosure-2024101116651.json](https://github.com/user-attachments/files/17351563/enclosure-2024101116651.json)

jasonspo, 2024-10-21:
> @jasonspo would you perhaps be able to give it a try with an increased insert clearance to see if that is indeed the issue? I have not experienced this issue before myself. Out of interest what size nozzle are you using and what printer?

Hi Bruce,Yes, increasing the insert clearance does fix the issue. I increased the wall thickness on the  last couple of designs I made, I'm wondering if that's contributing to the issue.  Hopefully it's an easy fix for you, I'm come to rely on this tool when I need a new enclosure for projects, it's awesome.  Thank you.  Jason

Jasperwolsing, 2025-03-09:
I have the same issue.

Did not had this like a year ago. Im going to try to increase clearance. Would be helpfull to share the clearance experience of others.

Flashforge Adventurer M5 
0.4 Nozzle

rmappleby, 2025-03-11:
Sorry, I'm new to 3D printing. How would I go about increasing the clearance, and what would that affect? I ask because except for the corners, the fit of the lid is excellent.

rmappleby, 2025-03-11:
Ok, sorry, worked it out. Feeling a bit sheepish now. Will try that on the next enclosure I print.

jasonspo, 2025-03-23:
It happened to me again. Is there a known reason and fix for this yet?  You can see the error from the hole to the weatherproofing between each piece.

jasonspo, 2025-03-23:
It is only a problem with the corners.  If they corners are cut out, the remainder of the weatherproofing fits.

[image]

dwertz1, 2025-06-06:
Same issue with lid not fitting.  Have not tried changing clearances.  Same on both of my printers, Longer LK5 and Bambu X1 Carbon.

[Temp and motion enclosure.json](https://github.com/user-attachments/files/20627608/Temp.and.motion.enclosure.json)

michmela44, 2025-09-06:
I have not printed anything or tried looking at the code yet, but I suspect many of the problems have to do with things like the "Base Hole Diameter" feature being added.

As a quick test, on a waterproof enclosure, if you set the Base Hole Diameter to something very large, it affects the shape of the base waterproof channel, but not the lid or insert.

[image]

I didn't check all of the different parameters, but any parameter that changes the shape of any 1 of these 3, has to affect the shape of all 3 for it to fit correctly.

Additionally, the Base Hole Diameter, I believe, was added to account for heat set inserts.

For that to work, the center of the lid holes and base holes have to remain aligned with each other, which also doesn't appear to happen. So any base hole increase/decrease, also needs to shift the lid hole over the center of the base

rmappleby, 2025-09-07:
Same here. Its a great tool and I'd be delighted to see it fixed so we can print good waterproof enclosures.

bruceborrett, 2025-09-09:
This is awesome @michmela44 , congrats on being the first contributor on this repo after more then 2 years of existence. I had a quick look at the changes and all looks fine, if someone else can just confirm that using your fork resolves their issues, then I will go ahead and accept the merge.

jasonspo, 2025-09-09:
> Ok. I haven't seen any recent activity from the author here.
>
> In the meantime, I published a version with my fixes here
>
> https://michmela44.github.io/easy-enclosure/
>
> You can try it there and see if it's better.
>
> I would suggest only printing slices of it first to test fitment to not waste filament. I have not tried a print myself yet.

Excellent!  Thank you, michmela44!  Many of us appreciate you taking the time and fixing this issue.   I loaded my files that had fitment problems before the fixes and they all work great now, thanks again!

michmela44, 2025-09-09:
> This is awesome [@michmela44](https://github.com/michmela44) , congrats on being the first contributor on this repo after more then 2 years of existence. 

No problem, it's the least I can do for the amount of time this will save me to get a waterproof case I need for a project at home.

I see that you've merged it now, so I will take down my published bersion so there's no confusion.

Thanks.

michmela44, 2025-09-10:
I'm not seeing the changes in the live version. Has it been deployed, or is it just behaving differently?